jasonwebb/tc-maker-4x4-router #11: Conversation Concerning Futre Home of CNC into CNC Land (I.e. Scary Bathroom)

XavierSockington, 2017-04-25:
Before homing or honing the machine, we need to have a conversation about the permanent home of this machine in CNC Land (i.e. Scary Bathroom) and what all is entailed in making that happen.

jasonwebb, 2017-04-25:
Can we tear down the wall to the bathroom to move the machine all at once?

Otherwise we'd need to take apart the machine to fit it through the door, which means we'll need to do a lot more work to recalibrate it, find new routing for all the tubing, run 220V, etc. But if that is what we have to do then better to get started sooner rather than later - like before we do anything else on the machine.

For the record, I'm on board with moving it before anything else is tackled, I just don't want to significantly add to the restoration timeline by moving it before the room is ready and solid plan for moving it is figured out. Definitely possible though, so I'll be making it my goal for the week to bring some clarity to this topic.

I'm unsure that Pete will be watching this repo - I'll try to ping him to find out what his plans are, and try to get the discussion going.

jasonwebb, 2017-04-25:
1. I've created a [new project](https://github.com/jasonwebb/tc-maker-4x4-router/projects/8) to track tasks to for this.
2. I've created a [new issue](https://github.com/jasonwebb/tc-maker-4x4-router/issues/12) that we can use to coordinate the discussion. 

I'm pretty new to using Github for team collaboration, and I didn't see a way to tie this issue to a project task, so I went ahead and made a relevant project task and generated a new issue for it. A little confusing, sorry!

Let's continue the discussion in the new issue so that it can be tracked!

#12

closed 2017-04-25 by jasonwebb