jasonwebb/tc-maker-4x4-router #12: Initiate discussion of relocation project - shop manager needs, member feedback, communication with board

jasonwebb, 2017-04-25:
Jon Alt (wood shop manager) has requested that we move (or develop a plan to move) the CNC machine into the adjacent "Scary Bathroom" space.

This would have clear benefits to both the wood shop and to the CNC machine, specifically:
1. Wood shop gaining much-needed floor space to better serve its members. 
2. CNC machine being isolated (better, at least) from wood shop dust.

Some of the challenges of moving the machine include:
1. Readying the bathroom to be capable of housing the CNC machine. Dust collection, power (120V and 240V), and evaluating practical space requirements for working with materials.
2. Moving the machine into the room. This is potentially the most crucial topic, requiring strong coordination and planning for how to get done. May need to disassemble the machine, reassemble, recalibrate in new space.

Given the complexity of the machine and the delicate nature of its calibration, we ought to figure this out before continuing on any further work, especially involving mechanical tuning and tweaking.

Jon and Pete need to discuss options, and Jason will be involved to facilitate discussion and help coordinate activity.

XavierSockington, 2017-04-25:
Send email to Key Players, that I know of, to initiate a meeting on this move.  Text of Email below:

To:
Pete, George, Alex, Jason (CNC aficionados)
David - (Facilities/Electrical)
Board Members (FYI that Discussion is happening)

This email is to get everyone on the same page, briefly explain why this conversation is happening and to set a meeting time when we can discuss this in person at depth (See Numbered Items at bottom of email for meeting time).

Being that we are close to getting the finishing room up and running, and the CNC is at a stage where - if we are to move the machines - now is the time - we need to figure out the best use of the Scary Bathroom.

The scary bathroom will lends itself to only 1 thing once we close it to finishing  - a dumping ground.  In order to avoid this - and to use our space to its fullest - I propose creating a new "CNC Land" for this space and moving all or most of the CNC into this room.

I know this is not a popular thing among the people working on the CNC - and I know it means a lot of work for people - but I think this is the best thing for both the organization, wood shop and CNCs as we grow (we're up to over 200 members now!)

1) All future Discussion/Tracking of this issue has been moved to a GitHub page at:
https://github.com/jasonwebb/tc-maker-4x4-router/issues/12

2) I'd like to propose a meeting on Sunday April 30th to Discuss this issue in person.  I know this is short notice - but If you could respond back with available times it would be appreciated.

3) During Quarterly Cleanup (May 6) I will be dedicating volunteers to cleaning out the scary bathroom, top to bottom, and closing the scary bathroom as a finishing room (it is nice enough outside for people to do finishing work there until the finishing room is up and running).

Thank you

jasonwebb, 2017-05-07:
From our meeting last weekend it looks clear that we will not be moving the CNC machine, so I'll go ahead and close this issue.

closed 2017-05-07 by jasonwebb